nanophotonics/nplab #6: seabreeze gui

hamidohadi, 2015-07-29:
a few buttons in the new gui don't work. For example: manual axes range buttons and saving spectra in particular. Otherwise, very helpful, thanks!

asanders, 2015-07-29:
These features were added to the GUI without being implemented yet in the actual code. Axes controls have not been added until a decision is made as to whether to stay with matplotlib or move to pyqtgraph, which already has these features. The datafile script in nplab is what enables data saving to a common file. I need to look into using these features to see how to implement saving spectra.

rwb27, 2015-09-30:
Manual limits now work.  Will need to check saving.

rwb27, 2015-09-30:
Saving also "works for me", I'll close the issue for now but reopen it if there's a problem.

closed 2015-09-30 by rwb27