nanophotonics/nplab #13: Camera crashing for particle tracking / long exposure

jm806, 2015-10-17:
The camera crashes randomly (at least seems so) at some point during the programme. Resetting the camera does not seem to work because cam.close() fails. It seems that there is not camera connection any more, i.e. the object "cam" is simply lost.
Trying now with recreating object "cam" without trying to close the camera...
Restarting the full programme did work so far.

rwb27, 2015-10-19:
We think we've sorted out at least some of this - the crashing on restart was because of the way gain/exposure are handled.  We now save gain & exposure every time a raw_snapshot is taken, and use this to reset the settings when the camera is restarted.  From 8125feb799a50ced181c1d910fbb2913a5a61148.

I've also merged in my logging improvements (7de59ac6f0dc01f81dfa36931091f294a6fa3fbe) and Jan will use this to save camera reset events to the HDF5 file.

rwb27, 2015-11-04:
Having swapped the camera in Lab 3, all appears to be well - it may have been an odd hardware issue, but it's strange that it still (mostly) worked in the Lumenera software.  There must be something I'm missing...

rwb27, 2016-06-03:
I think this is now a non-issue? I'll close it...

closed 2016-06-03 by rwb27