nanophotonics/nplab #28: Data browser window only with 'block=True'

jm806, 2016-06-02:
The data browser window disappears directly when 'block=False'. There is a reference of the GUI though. 
Using 'block=True' keeps the GUI alive

rwb27, 2016-06-03:
I can't reproduce this on my mac.  Odd.  Need to try on Windows...

rwb27, 2016-06-08:
I've reproduced this on Windows using the current master branch.  The problem goes away if I put the line `app = get_qt_app()` before popping up the GUI, I guess the problem is you need to hold on to the Qt application in order to avoid it getting garbage collected.  Quite why this has changed I can't understand!  Perhaps something updated in PyQt?  Either way, I'll make the code hold on to a reference to the Qt application, and hopefully that will fix things.

closed 2016-06-08 by rwb27

rwb27, 2016-06-08:
Actually, 4270372 fixed it.  I screwed up the global variable thing in the previous commit.

commit 6ef37eae (2016-06-08): Hold on to Qt application (closes #28)

There was an odd issue #28 where windows would immediately disappear.
I think this is due to not holding on to the Qt application object.
This is now done by default in `nplab.utils.gui.get_qt_app()` so the
problem should be gone.
commit 42703728 (2016-06-08): Qt app is now correctly retained

6ef37eaedc0bf27c488edf6968703f7f0a1eb748 looked like it fixed #28 but it
didn't.  This actually fixes it.