nanophotonics/nplab #31: Multiple inheritance with SerialInstrument

rwb27, 2016-06-14:
SerialInstrument is almost always combined with another Instrument subclass in multiple inheritance (one class for the communications bus, one for the logical instrument - camera, light source, shutter, whatever).  This leads to multiple inheritance which is confusing.

There is no particularly good reason SerialInstrument must be an Instrument, as it's almost always combined with an Instrument subclass anyway.

rwb27, 2016-06-14:
A possible solution would be to replace SerialInstrument with SerialCommunicationMixin (and do the same for VISA, etc. etc.).  These mixin classes would not inherit from Instrument - so the graph is simpler.  They would always be the second class in the inheritance list.

The issue then becomes one of making sure communications are properly initialised.  Perhaps we could standardise on a protocol:

open_communications() sets up the device, as is currently done in __init__
close_communications() closes it, as is done by close()

Instrument could even check for the existence of those methods and call them in its __init__method.  If we did that though, we'd need some flags to make sure that:

1. We don't call open_communications() twice by accident
2. We can avoid calling it from Instrument.__init__ if we don't want to.

YagoDel, 2020-10-28:
Anyone still interested in this? I've got the feeling Python3 forced us to patch quite a lot of the MRO issues, but I do like the idea of separating the communication protocol from the instrument

I've followed @rwb27 's idea and have the open_communications() method, and have the Instrument.__init__ detect for it's presence and call it

I've made a first pass that works with the SerialInstrument and VisaInstrument that I have in the lab, and if people are interested I'll put in another hour so that it's testable in other setups