nanophotonics/nplab #48: Traits and pyqt5

wdeacon, 2017-01-27:
It appears that traitsui has not been up dated to run pyqt5. In the latest version of spyder it does not appear to let you install pyside (I don't really understand why?) or in some cases it lets you install it but then it claims it cannot find it.

Any ideas rich? people are struggling to run the view_spectra.py file for looking at particle scanning data.

rwb27, 2017-01-28:
bother.  To be honest, we came to the conclusion a while ago that traitsui was on the way out.  I think the particle scanning is about the last place it's used, and my grand plan for particle tracking 2.0 was to use only pyqt.  I'm not sure how to fix it quickly, the right solution is just to stop using traitsui and that is not a quick fix.  I'll add it to my to-do, but I'm afraid I can't promise to sort it imminently...

wdeacon, 2017-01-30:
Thats okay Rich, if you would like I can try and rig something together quickly that would work even if it wouldn't be beautiful?

 On the side note of Particle tracking 2, I think within the next couple of weeks I am going to try and getting particle tracking going on my Rig so it may be a good time for me to tie together the code that you said you had sitting around? unless you think you are going to be able to find the time to work on it?

rwb27, 2017-01-30:
it depends on what people are using it for.  If it's just generating the summary HDF5 file (i.e. turning the one-folder-per-particle information into the big arrays people tend to analyse in Igor or whatever), then I could strip out the traits dependencies easily enough.  If it's analysing the data, we should probably just put together a nice plug-in for the data browser...

rwb27, 2017-01-30:
could you point me at the latest version of the analysis script people are using - I lose track...

YagoDel, 2019-12-12:
Seems traits does support PyQt5 [now](https://docs.enthought.com/traitsui/index.html#installation) the case

closed 2019-12-12 by YagoDel