nanophotonics/nplab #70: Bluesky

robmarkcole, 2018-08-13:
You might be interested in http://nsls-ii.github.io/bluesky/index.html
Python 3, asyncio, and hdf5 [exports ](http://nsls-ii.github.io/suitcase/)

rwb27, 2018-08-14:
Hello, that looks really interesting - I've actually moved on from the group in Cambridge and am slowly migrating my stuff to Python 3; it looks like bluesky addresses a number of the problems we've been trying to solve, so we'll definitely check it out here.

rwb27, 2018-08-14:
Sure - I imagine if we make use of it you'll be hearing from us one way or another!  I'm now at the University of Bath, starting a research group with a focus on open-source, automated instruments.

robmarkcole, 2018-08-14:
Congrats on the new position, sounds like a dream job. Do you have a website? In my spare time I'm a developer on https://www.home-assistant.io/ I'm interested in citizen science for e.g. air quality monitoring

rwb27, 2018-08-14:
thanks - it's a "not yet" on the website, I've written some content but not yet gotten round to putting it up anywhere!  Many of my projects are under my github profile though, and one of the most relevant things happening in the lab at the moment is LabDo, which you can find on @julianstirling's page

robmarkcole, 2018-08-14:
Another to checkout is https://github.com/ralph-group/pymeasure
Bit simpler than Bluesky

julianstirling, 2018-08-14:
Hi,
LabDo is far from useable now. I have big goals but not had the time to implement many of them yet. I will check out BlueSky when I have the time. LabDo has actually moved to https://gitlab.com/julianstirling/LabDo due to my tinfoil hat approach to anything Microsoft related.

asanders, 2018-08-14:
I've just received some notifications from this thread so thought I'd weigh in with some updates.

My efforts to upgrade nplab to Python 3.6 for general use outside of Nanophotonics and to keep it fairly lightweight are stored here:
[vipy](https://gitlab.com/asanders89/vipy/) is the base functionality of nplab
[datafile](https://gitlab.com/asanders89/datafile/) wraps h5py for data acquisition (previously part of nplab but quite useful as a standalone module)
[vipylib](https://gitlab.com/asanders89/vipylib/) is for storing specific VIs outside of the core functionality

It's not been updated in many months and it's unlikely I'll find a reason to continue developing these, at least not in the near future. The code is pretty much what I found to be the essentials from nplab with some extra functionality I added later. Some of it may be of use still.

robmarkcole, 2018-08-15:
You guys use github and gitlab, any reason why?

julianstirling, 2018-08-15:
I had everything on GitHub until Microsoft bought it. Microsoft have a history of buying functioning software and ruining it in the process of monetising it, or adapting it to make it subtly incompatible with known standards. I have moved every one of my own projects away from GitHub to GitLab, other projects are still on GitHub.