nanophotonics/nplab #74: solstis api

imanyakin, 2019-03-05:
Looks like m2 squared who make solstis lasers have updated their API so the code on light_sources/SolsTiS needs to be updated to reflect changes, in particular commands sent in the JSON - had to hack at it when working with Ermanno in tijmens lab this week. Looks like Omid/Optofluidics people have rolled their own version of code. We can ask them to integrate it to update

@YagoDel @Omid37 @ermannomiele

YagoDel, 2019-03-05:
I no longer have access to a SolsTiS, so I can really program this. I'm happy to provide support to understand what the code does though. 
I had a quick look to compare the nplab version and the optofluidics one, since I know Ana used my code to make theirs. I see no difference in the command structure. How did you hack it?

imanyakin, 2019-03-05:
Ok, well I'd be surprised if their code still works. Basically the "get wavelength" request was failing - we called up Msquared and they sent up an updated version of the manual/API - using that we had no problems getting control.

YagoDel, 2019-10-09:
I'm gonna guess that the changes @ermannomiele pushed in 5456f6403c1aad38827e32766c2f5e9665327d05 fix this issue

closed 2019-10-09 by YagoDel