nanophotonics/nplab #75: message_bus_instrument multiline ignore echo

YagoDel, 2019-06-10:
Currently:
`if self.ignore_echo == True: # Needs Implementing for a multiline read!
    first_line = self.readline(timeout).strip()
    if first_line == queryString:
        return self.readline(timeout).strip()
    else:
        print 'This command did not echo!!!'
        return first_line`
Would this fix it?
`if self.ignore_echo:
    first_line = self.readline(timeout).strip()
    if first_line != queryString:
        self._logger.warn('%s did not echo' % queryString)
        return first_line`

Need someone to test it with an instrument with multiline replies and echos

eoinell, 2019-12-06:
I tried it and i got 
[Instrument.ThorLabsSC10] - WARNING: ens? did not echo - 15:43

YagoDel, 2019-12-11:
Since there is currently **no** instrument that can test this feature (echoing + multiline), I went ahead and added the functionality anyway

commit 10c0e228 (2019-12-11): Fixes #75. Code cleanup
closed 2019-12-11 by YagoDel