nanophotonics/nplab #77: Merge jpg66 and wmd22 Thorlabs ELL18K stages

J-Griffo, 2019-06-21:
I have commited a lightweight class for the ELL18K rotation stage. This was adapted from code running locally in lab 5.

wdeacon committed a class for this a while ago (Named ELL8K.py. Assume this is a typo. If not and the stage version name has changed we are in the clear). I need to liase with him, as the stage has been updated with a new version (the original stages were essentially prototypes). We have 2 of the old stages and now 5 of the new ones.

Main changes noticed between the stages:
- The number of pulses to the motors to do a revolution has changed
- The stage no longer tracks when it has done a full revolution. ie. it's motor positions can go >360 degrees. If the position goes too high it throws an error

These are accounted for in my code but a merge is probably the best idea.

wdeacon, 2019-06-21:
Most probably a typo. This code was actually written by Iliya, not sure why its under my commits, but anyway.  Is this code going to break the old stages we have? if so is it possible to detect if the stage is the new/old styles and then can one run class that runs for both? or have you already implemented this? 

Thank for flagging this! and ten gold stars if you've already implemented this!

J-Griffo, 2019-06-30:
ELL18K.py now requests the number of pulses per revolution from the stage. 

Now just need to add in functionality\inheritance you would like @wdeacon :)