nanophotonics/nplab #87: Inheriting from object in python 3

eoinell, 2019-12-05:
Using 2to3/futurize on scripts does not affect the classes inheriting from object, however this creates an issue in when running in py3. Can new-style classes in py2 implicitly inherit from object?

YagoDel, 2019-12-06:
Python 2 cannot implicitly inherit from object. I think that was one of the main reasons driving the creation of Python 3 in the first place

Do you have a particular script that creates an issue? There are a few places in nplab that can lead to confusing MRO, and it might be the case we need to properly address it now that Python 3 has OO classes

eoinell, 2019-12-06:
https://github.com/nanophotonics/nplab/blob/d904fc84c334a56b8d33fdacf0ca20a827196f89/nplab/instrument/__init__.py#L29
I removed object from this line in python3 as it was throwing up an MRO error. It doesn't seem to make an error when running in python2. I've been able to run Lab 6 through this branch in python2, except the data browser won't render data. This is probably because of the comparisons of Nonetype and int.

closed 2019-12-06 by eoinell

eoinell, 2019-12-06:
Yeah, I removed it in python-3-from-future, and checked that it still works on py2 this morning.