nanophotonics/nplab #89: PyQt4 vs PyQt5

eoinell, 2019-12-05:
Throughout nplab, qt4 is used as default, and the matplotlib backend is often set to qt4_agg.
Should we be trying to switch to qt5 for the python3 branch?

eoinell, 2019-12-05:
https://github.com/nanophotonics/nplab/blob/d904fc84c334a56b8d33fdacf0ca20a827196f89/nplab/ui/hdf5_browser.py#L17

https://github.com/nanophotonics/nplab/blob/d904fc84c334a56b8d33fdacf0ca20a827196f89/nplab/ui/data_renderers.py#L9

Maybe I'm conflating the matplotlib backend with pyqt4/5

YagoDel, 2019-12-06:
There is a way of setting matplotlib to whatever Qt is installed (which is what I did in the commit)
<from matplotlib.backends.qt_compat import is_pyqt5>

However, I tried simply commenting those two lines out, and the test_data_browser.py runs in both Python 2.7 and 3.4 (although I only have PyQt5 in my environment)
So it might actually be that it's not necessary to manually set anything

YagoDel, 2019-12-06:
Fixed by 378fcf62a83073c68baa70a39eb877fa9f7b7b42

commit 378fcf62 (2019-12-06): Fixes #89 Manually setting matplotlib backend to available Qt environment
closed 2019-12-06 by YagoDel