nanophotonics/nplab #92: thorlabs_sc10 failing to query

eoinell, 2019-12-06:
https://github.com/nanophotonics/nplab/blob/a70c6d7b38b12315ec2cc98077737c6290b18710/nplab/instrument/message_bus_instrument.py#L120

Always prints ''this command did not echo!!!''
probably an encoding issue

Works fine in python 2.7.

YagoDel, 2019-12-07:
Might be related to #75 I'll try to have a look

wdeacon, 2019-12-07:
Have we tested many serial devices yet? The change to the default encoding to be unicode in python 3 should be problematic. So the write string may need a b to convert it, for example this should be b'mode?'

YagoDel, 2019-12-09:
I'm running multiple serial instruments (stage.SigmaKoki.HIT, temperatureControl.LakeShore.LS331, stage.apt_vcp_motor.DC_APT, spectrometer.Acton.SP2750...), so the base code is fine in Python 3

I think the issue is that there are only two instruments that use the ignore_echo option in [message_bus_instrument](https://github.com/nanophotonics/nplab/blob/a70c6d7b38b12315ec2cc98077737c6290b18710/nplab/instrument/message_bus_instrument.py#L130), and the problem with converting to Python 3 has to be in line 132, when comparing the strings.

I don't have any instrument that echos, but try replacing the current:
<
    first_line = self.readline(timeout).strip()
    if first_line == queryString:
        return self.readline(timeout).strip()
    else:
        print 'This command did not echo!!!'
        return first_line
>
with 
<
    first_line = self.readline(timeout).strip()
    print(first_line, queryString, first_line==queryString)
    if first_line == queryString:
        return self.readline(timeout).strip()
    else:
        print 'This command did not echo!!!'
        return first_line
>

That should tell you what the problem is. When you fix it, lets try and link it to #75 and close two issues in one

eoinell, 2019-12-10:
the first_line returned by 

def readline(self, timeout=None):
        """Read one line from the serial port."""
        with self.communications_lock:
            return self.ser_io.readline().replace(self.termination_read,"\n")
https://github.com/nanophotonics/nplab/blob/4b3a3508f6fa4feb2536ced465f742aa6e9fefd6/nplab/instrument/serial_instrument.py#L108

is an empty string

YagoDel, 2019-12-10:
`readline` should only return an empty string when there is nothing to read at the instrument end.
Three options for what should happen after sending `sc10.write('ens?')` (or some other command that requires a reply):

- **No reply**. Then I don't think the instrument is working. Does it work through the Thorlabs software?
- **Replies directly with `0` or `1`**. Then it is not an echoing instrument and you need to set `self. ignore_echo` to `False`
- **Replies first with something like `ens?`, and when you read again returns `0` or `1`**. Then it is an echoing instrument, and the issue is with the message_bus_instrument querying

closed 2019-12-10 by eoinell