nanophotonics/nplab #94: Package requirements

YagoDel, 2019-12-11:
Setting up a new environment takes quite a while atm
Having a list of tested packages and versions in the setup.py (or in the requirements.txt) would go a long way to make everything faster

eoinell, 2019-12-11:
This is what worked for Python 2 in lab 6.

conda install numpy scipy matplotlib future traits traitsui pyqtgraph pillow  h5py
conda install -c conda-forge opencv
conda install -c conda-forge pyserial
conda install -c conda-forge pyvisa
conda install -c conda-forge qtpy

(If using spyder 3, install latest version of spyder in Anaconda GUI. If using spyder 2
conda install -c anaconda spyder=2.3

eoinell, 2019-12-11:
I think for opencv in python 3 i ended up using pip however

eoinell, 2019-12-11:
There's bit of a difference between Lab PCs (e.g. pyserial, pyvisa) and laptops, but probably not enough to warrant different lists

YagoDel, 2019-12-12:
I was assuming people are not using environments, so I thought a requirements.txt for installation via pip would be the easiest.
However, traits does not directly install via pip (it requires binaries), so I'm thinking it'll be better to create a conda environment.yaml
Anything against this? Maybe @wdeacon has tried doing something like this on his MacOS

wdeacon, 2020-12-07:
I think a requirements.txt should do the job, I thought the conda install would also use requirements.txt? if this is already complete can we close the issue?

YagoDel, 2020-12-22:
It ain't fully complete, but I stopped working on it since I've stopped setting up different Py2 and Py3 environments (I use Py3 for almost everything now, and haven't updated the Py2 computer in a long time, since it just uses a tiny subsection of the code).
I think it basically just needs some heavier testing