nanophotonics/nplab #95: PyQt overloaded signals

YagoDel, 2019-12-11:
I'm getting a `KeyError: 'there is no matching overloaded signal'` in this [line](https://github.com/nanophotonics/nplab/blob/6a191088739e35d7ba8ab28fa38327ebcdb7ea67/nplab/instrument/stage/__init__.py#L246) of StageUI, when running in Python2, both PyQt4 and PyQt5

Following [this](https://stackoverflow.com/questions/11902109/overloaded-pyside-signals-qcombobox), it works if I change `step_size_select.activated[str].connect(partial(self.on_activated, i))` to `step_size_select.activated[unicode].connect(partial(self.on_activated, i))`, but then this fails in Python 3

I don't understand this issue, since clearly people are using this code in Python 2. Can anyone reproduce this KeyError? I'm using qtpy as the PyQt wrapper.

eoinell, 2019-12-11:
I've asked around, and I'm not sure that people here use the stage UI. Most just use CWL, or the joystick in each lab.

YagoDel, 2019-12-11:
😂 🤦‍♂ 
Had a look through usages of Stage, and found the following classes that use the StageUI, as well as current people that have used/modified them (semi-)recently:
- @DemelzaW: pi_stage.py
- @imanyakin: Thorlabs_NR360SM.py
- @wdeacon: xyzstage_wrapper.py, prior.py, Piezoconcept_micro.py, thorlabs_kinesis.py

eoinell, 2019-12-12:
Demelza doesn't, Ilya does, but isn't sure whether he inherits from stage UI

imanyakin, 2019-12-12:
Ah, I didn't realize it's this one - I haven't used this in a long time and don't intend to.

YagoDel, 2019-12-16:
Since people aren't currently using it, and it works fine in Python 3, we will not provide Python 2 compatibility for this

closed 2019-12-16 by YagoDel