nanophotonics/nplab #97: ViewBox of graphs in preview_widget of Camera base class extends beyond widget edges.

eoinell, 2019-12-14:
This occurs in python 2 and 3, but_only_ on the **python3** branch. 

The example I've seen is using the Andor:
When you take a single-track spectrum, the x-axis extends beyond the edges of the widget such that it's no longer possible to see the values on the y-axis.

YagoDel, 2019-12-16:
Screenshot please 🙇

eoinell, 2019-12-16:
[image]
[image]
[image]
[image]

Each is a before/after image pair of clicking capture in single track mode. (the 1st gui hasn't set the x_axis)

eoinell, 2019-12-16:
I made a dummy class just there of CamerScaledRoi which replicates the issue; it works in master but not in python3

eoinell, 2019-12-16:
Side-by-side:
[image]

closed 2019-12-16 by eoinell

YagoDel, 2019-12-17:
For anyone looking at this, the fix is [line 204](https://github.com/nanophotonics/nplab/blob/25713c882112a3d0a765b00c9d4e5bad6d95df7a/nplab/instrument/camera/camera_scaled_roi.py#L204) in 25713c882112a3d0a765b00c9d4e5bad6d95df7a