nanophotonics/nplab #100: Seabreeze GUI won't plot if processed spectrum contains NaNs

mjh250, 2019-12-17:
The "process_spectrum()" method in "instrument.spectrometer" replaces infinities with "NaN" in processed spectra (after subtracting the background and dividing by the reference). This stops the spectrum from being plotted in the GUI when using the latest version of pyqt.

eoinell, 2019-12-18:
Just a matter of this before plotting?
```
def NaNs_to_zeros(array):
        '''converts all NaNs in a multidimensional numpy array to 0'''
        return np.reshape([0 if np.isnan(i) else i for i in np.flatten(array)], np.shape(array))

```
https://github.com/nanophotonics/nplab/blob/78ba3dd27b931a5b652e7440f404c7ee675a90d4/nplab/instrument/spectrometer/__init__.py#L652 
`spectrum = np.array([NaNs_to_zeros(s) for s in spectrum]) # in case the spectra are different lengths
`

wdeacon, 2019-12-18:
That will certainly work but I don't know if it's the best approach as it makes it look like there's no light when that is not the case. It means there's no light in the reference (most likely) so I'd suggest it's better to delete the point. 
 wavelengths = wavelengths[~np.isnan(Spectrum)]
Spectrum =Spectrum [~np.isnan(Spectrum)]

YagoDel, 2020-01-08:
Seems like a bug in pyqtgraph itself https://github.com/pyqtgraph/pyqtgraph/issues/1057

If you can't ignore it for now, I'd say go with an earlier PyQt (thread says 5.13.0 works)

eoinell, 2020-01-10:
Still hasn't been tested for multiple spectrometers as I don't have any rigs that use them, but works for 1 spectrometer

closed 2020-01-10 by eoinell