nanophotonics/nplab #107: Andor's crop mode crashes python

eoinell, 2020-01-16:
clicking the 'crop' mode on the Andor gui crashes python

eoinell, 2020-01-16:
the set_image function appears to be deprecated
```
            self.Andor.set_image(1, maxy, maxx, current_binning, current_binning)
        else:
            self.Andor.set_camera_parameter('IsolatedCropMode', 0, maxy, maxx, current_binning, current_binning)
            self.Andor.set_image()

```
what did it do?

YagoDel, 2020-01-17:
Have a go now

eoinell, 2020-01-17:
The crop mode now works thanks! The roi mode however crashes:
[image]
I'm not clear on what the difference between the two modes should even be tbh

YagoDel, 2020-01-17:
Did it crash before or after you made this commit 1184f81d9e3f6a6228c6dd0edd61c2e479caacbd? I ask because self.detector

Cropping changes the effective size of the detector (changing self.detector_shape as well). I think it's intended for when part of the CCD is not in the optical path. There is a slight difference in how the electronics readout the CCD charge (changing things like speed and noise level). And in principle you could crop the CCD, and then use a ROI on top of that, but I haven't figured out how to make a GUI for that, so it's not implemented.
For most purposes, the only difference is that one of the cropping vertices is always the top-left, whilst ROI can happen in any rectangle

eoinell, 2020-01-17:
After. 

But the crop checkbox caused python to crash before I did this because the `detector_shape` attribute was NoneType.

YagoDel, 2020-01-17:
Weird. I'll have a look Monday (obvs not in the lab anymore today). It'd be helpful if you check if it works before that commit

eoinell, 2020-01-17:
Yeah it was that commit that was breaking it. I just changed detector shape to 1 and now neither errors appear :)

closed 2020-01-17 by eoinell