nanophotonics/nplab #112: set TEC temperature on OO spectrometers crashes out

imanyakin, 2020-01-30:
Running GUI, this will block the command line until the window is closed.
Traceback (most recent call last):
  File "C:\Users\im354\workspace\nplab\nplab\instrument\spectrometer\seabreeze.py", line 474, in gui_set_tec_temperature
    self.spectrometer.set_tec_temperature(float(temp_value))
TypeError: 'int' object is not callable

eoinell, 2020-01-30:
Python 2 or 3?

eoinell, 2020-01-31:
86dc8deecc5b67a2f841db888bcb1e806c81ac0e 
I changed how the gui and the temperature interact so that reading the temperature doesn't set the temperature to the read value. The GUI works fully.. i've reimplemented the property anyway so hopefully any old code will work again

closed 2020-01-31 by eoinell