nanophotonics/nplab #117: Andor's ExtendedImageView replaced with blank box when resizing window or changing capture mode

eoinell, 2020-07-22:
When using the Andor GUI the image viewbox turns black when the gui is resized, or when switching to FVB or another spectrum acquisition mode. This is reproducible in part with the DummyCameraRoiScale class in camera_scaled_roi, when resizing the image pops in and out of view.

YagoDel, 2020-07-28:
Versions for PyQt and pyqtgraph? (I assume this is Python 3 on Windows?)

eoinell, 2020-07-28:
qt=5.12.5, pyqt=5.12.3, pyqtgraph=0.10.0

Yes, py3, win10. 

We did the python3 update at the same time everyone upgraded to windows 10, I guess if people are still using older versions they're probably using the python2 branch

YagoDel, 2020-07-28:
There's a new pyqtgraph (which is the one that changed the quickMinMax)

But that doesn't change the issue here (I've got qt=5.9.7 and pyqtgraph 0.11.0), which at least I've managed to reproduce. It feels like a pyqtgraph bug...

eoinell, 2020-07-28:
Agreed it's probably pyqtgraph, although what I don't understand is why it happens here, but not in Spectrometer's ImageView or Camera's.

YagoDel, 2020-07-28:
I think the problem is that `CameraPreviewWidget` is subclassing `pygtgraph.GraphicsView` and `DisplayWidgetRoiScale` is subclassing `pygtgraph.ImageView` (that's the only difference I can see right now). And since then the code in the background is updating things and assuming they are the same type of object, I guess it's messing it up

closed 2020-07-29 by YagoDel

YagoDel, 2020-07-29:
In the end it was a signal and queuing problem

commit d6e4f6b6 (2020-07-29): Fixes #117 pyqt queuing issue fixed by following what is already done in camera.__init__