nanophotonics/nplab #124: Instrument settings

YagoDel, 2020-10-14:
My problem: I kinda want the Andor to go back to the settings I last left it in, not change when `self.initialize` gets called

More general problem: have instruments save/load a set of settings. Could be handled similar to metadata_property_names, maybe even the same actually

rwb27, 2020-10-14:
Related idea I never managed to implement: have a context manager that allows you to temporarily change a property.  So instead of:
```
prev_exposure = camera.exposure_time
camera.exposure_time = 10
image = camera.take_snapshot()
camera.exposure_time = prev_exposure
```
you could do:
```
with camera.temporarily_set(exposure_time=10):
    image = camera.take_snapshot()
```
That is not only way neater (especially if you're changing more than one setting), it's also much more robust to errors, etc.

Consistently managing settings is tricky - so I will be curious to see what you come up with here.  I've quite enjoyed our settings mechanism in [LabThings](https://github.com/labthings) because a side-effect of the way we've done it is that it becomes trivially easy to download all the current settings as JSON, and restore the state of the instrument by sending said JSON back to the server.

YagoDel, 2020-10-15:
Related to that context manager, I had a quick go adding this as an `Instrument` method and tested it with the Andor to do what you described:
```
@contextmanager
 def temporarily_set(self, **kwargs):     
    try:         
        original_settings = dict() 
        for key, value in kwargs.items(): 
            original_settings[key] = getattr(self, key)
             setattr(self, key, value) 
        yield original_settings 
    finally: 
        for key, value in original_settings.items():
             setattr(self, key, value)
```

Thanks for pointing out LabThings! I was indeed thinking something with that kind of JSON functionality (although I was thinking YAML), but indeed finding something consistent is going to be interesting...

YagoDel, 2020-10-15:
Made a first pass at this (in branch [issue124](https://github.com/nanophotonics/nplab/tree/issue124)). It works for what I wanted to do with the Andor