nanophotonics/nplab #135: ImageView

YagoDel, 2021-02-26:
The way the current ExtendedImageView handles percentiles is at best spaghetti (thanks to yours truly). At some point, pyqtgraph also changed it's quickMinMax signature, which started breaking the spaghetti.

Me and @mjh250 have tried patching it, but it seems we're testing it on different machines so my code breaks his and his breaks mine.

I think I can fix it, but I need some testing from @mjh250

eoinell, 2021-03-02:
I think I've fixed this in the py38 branch! It's a bit of a mess though for sure.. quickMinMax used to return a tuple, and now it returns a list of tuples

eoinell, 2021-03-02:
Oh yeah and I tested it it in machines that used the older and newer versions of pyqtgraph.

YagoDel, 2021-03-02:
Thanks for testing it!
So when you said you tested it with different pyqtgraph versions, you mean you tested the issue135 branch or the py38?

eoinell, 2021-03-02:
py38! I've opened a pull request for it

YagoDel, 2021-03-02:
I ask because in py38 you fixed it inside the getProcessedImage method, but issue35 I think completely removes the modification, so makes it less spaghetti

eoinell, 2021-03-02:
That sounds much better - but currently it doesn't work on older pyqtgraph versions is that it? or not sure?

YagoDel, 2021-03-02:
I'm not sure. I would have thougth that if pyqtgraph is backwards compatible, then it should work. But I have not tested it

YagoDel, 2021-03-03:
Quick note on this: running `camera_scaled_roi` in `master` currently doesn't work in Python 3.7, pyqtgraph 0.11, PyQt5